walkingmachinegraveyard/sara_commun #7: Error when using roslaunch

salco, 2016-09-29:
Probleme en lien avec les *.launch

JeffCousineau, 2016-09-29:
Pas full clair... quel launch ? :P

salco, 2016-09-29:
C'est en cours de checkup je vais lister le stock.

salco, 2016-09-29:
Voila jai passe a travers tout les .launch de sara_commun/sara_launch/ .

gregoirefra, 2016-10-18:
@salco est-ce que tout les launch fonctionne a nouveau? la structure des fichier est en train d'etre discuté dans #33

JeffCousineau, 2016-10-18:
Tous les launch devrait pas marcher et c'est normal :P 
Parce que certains launch file ont des dépendances !

gortium, 2016-10-18:
Ils y en a beaucoup qu'on a juste arrêté de se servire et qui sont devenue désuet. Il faut discuter aussi de l'emplassement des launch file. Dans quelle cas on met le launch dans sara_launch ou dans quel cas on le met dans le package lui meme. Je crois que je vais faire une flowchart pour l'installation de package et la creation de launchfile xD

gregoirefra, 2016-10-19:
Please see #35 to continue the discussion

closed 2016-10-19 by gregoirefra