walkingmachinegraveyard/sara_commun #17: mouvement bras teleoperation

gregoirefra, 2016-10-05:


Philippe117, 2016-11-22:
Pour commencer, ça serait important de bien documenter wm_arm_driver.
Sans documentation, s'est presque impossible de travailler.
J'ai fait une issue dans sara_robot à ce propos.

gortium, 2016-11-28:
Le truc, c'est que le gars qui la écrit n'est plus dans le club.. donc faut passer a travers le code et le comprendre.. et oui le documenter  (super important)!! Prochaine session je vais m'attaquer a ros_control et je vais intégrer le contrôle des drives des roue, mais aussi celle du bras donc je vais avoir besoin de passer au travers du code tôt au tard. Si tu peux essayer de comprendre wm_arm_driver le plus possible d'ici la ce serrais vraiment bien.

Philippe117, 2016-11-28:
T'en fait pas, je m'y suis attelé et sa commence à bien rentrer dans ma tête. J'ai déjà commencé à rédiger la doc.
https://github.com/WalkingMachine/sara_robot/wiki/wm_arm_driver

gortium, 2016-12-04:
done

closed 2016-12-04 by gortium