walkingmachinegraveyard/sara_commun #20: Choix de routeur

gregoirefra, 2016-10-05:


salco, 2016-10-07:
C'est les contrainte et objectif de ca? A ton reelment besoin d'inclure un routeur?

gregoirefra, 2016-10-18:
@maximest-pierre est-ce qu'on a un update la dessus? Si je me trompe pas l'ordi est confirmé donc il faudrait commencé a accélérer le choix du router et switch

maximest-pierre, 2016-10-18:
j'ai trois ou quatre routeur sur mon bureau en attendant pis j'ai aussi un routeur en tete de microtik.

salco, 2016-10-19:
Moi je réitère ma position de séparé switch et router. Et de choisir qqch de plus léger en consommation et coût. eg. (samr21 or Samd21)+wifi+ ethernet(30$+30$+30$digikey), raspberry-pi3(with wifi)(67$digikey), worst case(beaglebone + dungle_wifi( j'ai jamais expérimenter avec beagle au contraire des deux autre proposition))(16.99$ http://www.diygadget.com/802-11b-g-n-nano-usb-wifi-adapter-dongle-with-external-antenna-for-raspberry-pi-ralink-rt5370.html)

gregoirefra, 2016-10-19:
Est-ce qu'on pourrait avoir les avantages et désavantages de vos options chacun @maximest-pierre et @salco avec une liste de pièce exact, de cette façon je pense qu'on pourrait avoir une meilleur idée de la situation et mieux prendre la décision plus que "yen a trois ou quatre sur mon bureau"

gregoirefra, 2016-10-19:
Et avant tout voici les requis du tout : 

- Connecter tous nos devices réseau ensemble donc 3 connections si je me trompe pas
- Avoir un accès wifi au robot et/ou permettre au robot de se connecter a un wifi
- Avoir une demande de power le plus minimal possible
- Avoir une emprunte physique la plus petite possible pour ne pas prendre trop de place dans le robot

salco, 2016-10-19:
1. (samr21 or Samd21)+wifi+ ethernet
90$ + (30-130$ pour la switch) 
Avantages
-  Wifi et etheternet, code deja fait avec ASF de atmel
-  Module dédier au traitement du wifi
-  Module Petit (1/2pied^2 environ)
-  Low-power
Désavantages
- Kit de développement donc idéalement on doit pas rester avec ça car on peut le réduire et optimiser la taille.

2. Raspberry-pi3
67$ + (30-130$ pour la switch) 
Avantages
-  Wifi et etheternet integré
-  Linux Debian ou autre peut etre mis dedans
-  Module Petit (1/2pied^2 environ)
-  Peut remplacer un boad actuel équivalen ou permetre l'ajout dun nouveau node pour de futur projet(plusieur I/O disponible sur un header)
-  Devlopement rapide
Désavantages
- Linux, On doit faire l'integration de ros ou autre au besoin(le cas ou il serai un node pour le futur)

2. beaglebone + dungle_wifi
27$ + (30-130$ pour la switch) 
Avantages
-  Wifi grace a un dungle usb
-  Linux peut être mis dedans
-  Beaglebone déjà existant alors pas de grosse modification requise
Désavantages
- Linux, On doit faire l'integration de ros ou autre au besoin(le cas ou il serai un node pour le futur)

JeffCousineau, 2016-10-19:
J'ai un dongle wifi sur mon pi, on testera quand on aura le beagle !

gortium, 2016-10-19:
@salco aucune de tes options ne comporte un minimum de 3 ports ethernet. Tu y rajouterais une switch?

salco, 2016-10-19:
@gortium  Je specifie dans chaque option  (30-130$ pour la switch), alors oui je prevois rajouter une switch et sa permet de séparer les deux section.

gortium, 2016-10-19:
Oups sorry je n'avais pas vue ^^

maximest-pierre, 2016-10-19:
Router1 : https://routerboard.com/RB953GS-5HnT-RP (169.99$)

Router2 : https://routerboard.com/RB2011UiAS-2HnD-IN (129.99$)

salco, 2016-10-20:
Le choix qui sera fait va influencer directement l'issue #28 .

closed 2022-12-14 by gregoirefra