walkingmachinegraveyard/sara_commun #32: Ménage répo

gortium, 2016-10-17:
Plusieurs package ne sont pas dans les bons répo..

- [ ] roboteq

- [ ] wm_arm

- [ ] wm_hand

- [ ] wm_mecanum_cmd

- [ ] wm_odometry_feedback

Ceux-ci devraient être dans sara_robot

JeffCousineau, 2016-10-17:
en ce moment il n'y a absolument rien par contre dans sara_robot

gortium, 2016-10-17:
Justement haha xD Le but est de mettre tout se qui est hardware que tu aurais pas de besoin sur ton ordi en simulation

JeffCousineau, 2016-10-17:
Gud essaie de détailler chaque repo ce qu'il doit contenir sur le wiki !

gregoirefra, 2016-10-17:
Hmm question comme ca est-ce qu'il y aura pas une meilleur division possible des repos? J'ai l'impression qu'on pourrait faire des submodule git. Oui ca ferait des tones de repo dans l'organisation mais ca permettrais de cloner juste les packages necessaires quand on en a besoin

gortium, 2016-10-17:
Ouais c'est effectivement ce que font beaucoup de team! Ca permet un réutilisation plus facile aussi (robot kinova ;P) !! Mais ca va rendre le tracking des issue plus dur oO de la l'utilité des Project? Je sais pas

gortium, 2016-10-17:
Le compromis que j'avais choisi étais de faire un repo par "user potentiel" si on veut.

gortium, 2016-10-28:
Merged.

PR #44 (2016-10-28): [DONE]Feature multiple repo
### TODO:

- [x] Modify sara_install to reflect change here

### Step to merge:

1. Merge the sara_install branch of the same name.

2. Merge the sara_robot branch of the same name.

3. Merge this branch.

4. Pull change on desktop and Sara

5. Modify dependencies.rosinstall to fetch the develop branch (do the same when merging to master)

6. Build

Everything should build as usual. Package moved were obviously not needed to build or simulation. 

Now desktop should only pull sara_commun and sara_simulation if needed

Robot computer should pull sara_commun and sara_ [... cut]
closed 2016-10-28 by gortium