walkingmachinegraveyard/sara_commun #33: Launchfile de téléopération

gortium, 2016-10-18:
Dans l'optique d'avoir un simulation qui fonctionne, les launch file de téléperation ne devrait pas contenir les launch des drives. La téléopération devrait être indépendante du hardware et devrais donc fonctionner en simulation aussi.

maximest-pierre, 2016-10-18:
Tu veux dire ceci 
```
<launch>
	<include file="$(find wm_static_tf)/launch/wm_static_tf.launch"/>
	<include file="$(find wm_mecanum_cmd)/launch/wm_mecanum_teleop.launch"/>
	<include file="$(find roboteq_driver)/launch/wm_driverLauncher.sh"/>
</launch>
```

changer a sa 

```
<launch>
	<include file="$(find wm_static_tf)/launch/wm_static_tf.launch"/>
	<include file="$(find wm_mecanum_cmd)/launch/wm_mecanum_teleop.launch"/>
</launch>
```

gortium, 2016-10-18:
Hmm roboteq c'est la pince, ca ne devrais pas être la non plus. Je parlais surtout de wm_mecanum_cmd et du fait qu'il faut partir la node des drive quand tu est en simulation.

maximest-pierre, 2016-10-18:
robotiq ses pince roboteq ses drive mais je comprend quoi tu veux dire.

gregoirefra, 2016-10-18:
Rendu la ca serait pas mieux que la sim launch toute et que roboteq et robotiq lance le teleop aussi

JeffCousineau, 2016-10-18:
ouais ben le launch file de sim vas lancé le teleop, mais on veut pas que teleop lance les drive and shit !

gortium, 2016-10-18:
Il devrait y avoir sara_configuration(drive, pince sont dedans), sara_simulation, puis tout les autre launch file devrait être unitaire.

gregoirefra, 2016-10-18:
Un dessin pourrait etre bien pour faire ca correct et l'issue #7 est relier a la problematique ici aussi je pense bien

gregoirefra, 2016-10-19:
Please see #35 for the rest of the discussion

closed 2016-10-19 by gregoirefra