walkingmachinegraveyard/sara_commun #35: Launchfile structure

gregoirefra, 2016-10-19:
This issue is to merge two different conversion in one. #7 & #33 

As it was noted on the previous issue, the launch file structure and logic is getting really messy and some have incorrect path to other launch file since they are no longer maintain. 

The new structure and logic with the launch file should be discuss here so we can propose something better.

maximest-pierre, 2016-10-19:
I propose a package that contains the launch file for the different module like sara_launch.

gregoirefra, 2016-10-19:
Could you describe more your proposition @maximest-pierre. Right now I understand you do not want to change anything and propose we keep everything the same. Is it right? 

The issue first come's from the fact that the teleoperation launch was also launching the drive for the wheel while teleoperation could also be used in the simulation, so an exemple was given to reverse the process that launching the simulation would launch the teleoperation and launching the drive would launch the teleoperation.

maximest-pierre, 2016-10-19:
Keep the main launch file inside one package for easy use. Like start navigation, start teleoperation no sim, gmapping. We can make a subfolder for simulation launch file.

gortium, 2016-10-19:
Ok I think I got it.
Because there is package installed (no source), package cloned (no modification possible), and our package, I think that it would be simpler to put every launch file in Sara_launch. But they need to be separated in main launchfile (launchfile calling launchfile) and unitary launchfile (launchfile starting one node)

gortium, 2016-10-21:
Je ne sais pas encore quoi faire des remote launch file par example.. 0o Dans sara_simulation ca fait pas de sens pour quelqu'un qui ne travail jamas avec le robot.. et dans sara_commun ca fait pas de sens parce qu'il vont se retrouver sur sara..

gortium, 2016-10-23:
Un autre option serrait de nous obligé a toujours spécifier le paramètre "machine" dans les launch file unitaire. De cette manière, les main launch file remote sont les même que  ceux sur sara et peuveuve être placé dans sara_commun/sara_launch

closed 2022-12-14 by gregoirefra