walkingmachinegraveyard/sara_commun #37: Alternative de gmapping

maximest-pierre, 2016-10-20:
Dans le code, il y a de la reference a rtabmap qui est un algorithme de slam. Il a l'aire a mieux fonctionner dans une environnement plus ouvert sans reference et créer pour des camera kinect et kinect clone. 

http://wiki.ros.org/rtabmap 

Les alternative sont hector slam et gmapping (actuellement utilisé)

gortium, 2016-10-20:
Yep j'avais fait plusieurs test avec rtabmap et ça semblais prometteur mais trop incertain pour la compé.. Ce serrais le bon temps de remettre ce projet sur les rails ! 
En plus ça vient de l’université de sherbrook, donc facile a contacter si besoin est !

maximest-pierre, 2016-10-20:
Cool je vais essayer de faire des test avec la kinect et la xtion quand on aura le temps.

JeffCousineau, 2016-10-20:
ça vient de fucking FRANCOIS MICHAUD !! ENCORE LUI TBK @red1laref