walkingmachinegraveyard/sara_commun #38: sara configuration au lancement de l'ordi

maximest-pierre, 2016-10-20:
Une chose qui serai intéressant sa serai de rouler sara_configuration quand il démarre.

gortium, 2016-10-20:
Faut juste s'assurer de rien launcher qui pourais la faire bouger haha juste des driver genre

salco, 2016-10-20:
Genre un shell qui s’ouvre au startup? ex: http://stackoverflow.com/questions/12973777/how-to-run-a-shell-script-at-startup

salco, 2016-10-20:
Il faudrait un launch file modulable ou un minimal et complet genre.

gregoirefra, 2016-10-20:
Hmm je serais plus d'avis qu'on aille genre un control panel web qui s'ouvre pour facilement launch les différentes choses parce que sinon ouais faudrait limiter grandement ce qu'on launch au démarrage

gortium, 2016-10-20:
Dans sara_configuration, normalement c'est juste le minimum. Pas de nav, pas de moveit, pas rien qui bouge. Juste le urdf, les different driver pour etre pret a lutilisation. Apres, la on launch des truc specifique avec l'ecrant.

gortium, 2016-10-20:
Et pour le start up scrip on a deja quelque chose compatible avec ros. Moi et @JeffCousineau on avais deja fait des tests avec ce package: https://github.com/clearpathrobotics/robot_upstart
Je crois meme qu'il est deja dans l'instalation de base de sara xD

maximest-pierre, 2016-10-21:
Le script de sara_configuration demarre gmapping.