walkingmachinegraveyard/sara_commun #46: Vélocité des moteurs

maximest-pierre, 2016-10-21:
Vérifier la vitesse nécessaire pour pogner la tractions dans plusieurs type de plancher.

- [ ] Ciment
- [ ] Céramique 
- [ ]  Tapis

gortium, 2016-10-21:
ROS et les drive sont sensé s'adapter pour atteindre la vitesse désiré peux importe le sol non?

maximest-pierre, 2016-10-21:
Des fois il a des command cmd_vel mais le robot bouge pas. Mon idee est de faire des launch file different pour les divers surface. vel_y_min = 0.2 pour ciment pis vel_y_min = 0.4 pour tapis.

JeffCousineau, 2016-10-21:
Est-ce qu'il y a un minimum de vitesse pour que le robot avance, genre setté dans les drives ?

maximest-pierre, 2016-10-21:
Dans move_base ya un parametre pour sa

gortium, 2016-10-21:
En fait les cmd_vel ne devrait pas changer d'apres la surface ! une vitesse est une vitesse. Ce qui va changer c'est l'effort (N) pour arriver a cette vitesse. C'est une autre raison pourquoi je veux utiliser ros_control. Il permet de controler l'effort par joint.

JeffCousineau, 2016-10-21:
Ouais tu devrais pas avoir à dire au robot quel type surface il roule, dans une maison t'as autant du tapis que du plancher

gortium, 2016-10-21:
Hmm ros_control ne fait pas de la magie non plus, il va pt falloir creer des service pour dire a ros_control quel suface sara est dessus.. et un moyen de detecter ca.. va falloir penser à ça.. mais l'idée de base c'est d'avoir le meme cmd_vel peux importe, et puis dans ros_control, choisir l'effort qui convien.

gortium, 2016-10-21:
Ha je me trompe pt..! il y a des controlleurs dans ros_control.. c'est pt eux qui augmente l'effort automatiquement.. hmmm a investiguer. Va pt falloir creer notre propre controller a partir de [wm_mecanum_cmd](https://github.com/WalkingMachine/sara_commun/tree/master/wm_mecanum_cmd)

gortium, 2016-11-30:
Je close parce que l'issue #8 (ros_control) va s'en occuper.

closed 2016-11-30 by gortium