walkingmachinegraveyard/sara_commun #51: TTS implementation

JeffCousineau, 2016-10-22:
- Present different tts solutions with pros and cons.
- The best solution and the best voice that sound natural.
- It needs to be able to speak english and french. It's possible to use 2 differents package for the 2 langages. 
- It's also possible to use an online solution but an offline alternative is need in case there's not internet.

Possible packages : eSpeak, pico2wav, festival, mbrola, external sound card...

maximest-pierre, 2016-10-23:
I would go with google tts api and festival with a custom voice.

Google tts sounds really human and natural but needs an internet connection.

Festival is capable of doing lots of cool stuff but needs a voice model.

gortium, 2016-10-24:
Can we use a internet connection for the competition ?

gortium, 2016-11-03:
Ca marche ? on close ?

JeffCousineau, 2016-11-03:
on l'a pas fait ! :P

gregoirefra, 2016-11-03:
Est-ce qu'il y a une Pull request qui adresse l'issue?

gortium, 2016-11-03:
Ok c'est pas l'implementation de tts alors, c'est plus une etude exploratoire