walkingmachinegraveyard/sara_commun #73: Firefox bug d'affichage

TheDaringDan, 2016-11-18:
[image]

Il semble avoir une case vide qui apparaît quand il y a un sous-titre donné a un membre.

JeffCousineau, 2016-11-18:
To reproduce the same error :

1. Open firefox in linux
2. Open the menu and click on devopper tab
3. Click on "responsive design mode"
4. Select "1920x900" preset

gortium, 2016-11-30:
Cette issue ne devrait pas etre ici. Elle devrait etre dans le repo du site.

gortium, 2016-12-01:
Issue moved to [WalkingMachine/walkingmachine.github.io #48](https://github.com/WalkingMachine/walkingmachine.github.io/issues/48) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-12-01 by gortium